foxglove/foxglove-sdk #7: Rename repo to message-schemas

defunctzombie, 2022-01-17:
**Description**
https://github.com/foxglove/ros-message-schemas/blob/6807f46716017a0461656f56dde5274943c7d528/LocationFix.proto

This repo contains a non-ros proto file which corresponds to our documented schema: https://foxglove.dev/docs/messages/location-fix

We should rename this repo to `message-schemas` and have to top level folders under proto: `ros` and `foxglove` where we store the two variants.

defunctzombie, 2022-01-17:
@jtbandes thoughts?

jtbandes, 2022-01-18:
Ok. It seems useful to have both the JSON Schema and .proto definitions available for our custom schemas as well as the ROS schemas. Don't think those will be able to share any of the same generation scripts (since the script uses parsed .msg files as input) but we could keep them in the same repo.

defunctzombie, 2022-03-21:
Renamed

closed 2022-03-21 by defunctzombie