foxglove/foxglove-sdk #18: Publish foxglove_msgs package for ROS 1 + ROS 2

amacneil, 2022-05-07:
Need to publish a `foxglove_msgs` package for ROS 1 (melodic/noetic) and ROS 2 (foxy+).

Bonus points if we can avoid creating a new repo for this, and delete https://github.com/foxglove/ros_foxglove_msgs in the process (I can live with a separate `-release` repo if that is unavoidable).

Maybe that requires moving the ros messages under subfolders like:

- `schemas/ros1/foxglove_msgs/msg/Grid`
- `schemas/ros2/foxglove_msgs/msg/Grid`

Rosbridge seems to work fine as a multi-package monorepo, so we should see if we can copy however they do it.

jtbandes, 2022-05-18:
Currently blocked on https://github.com/ros2-gbp/ros2-gbp-github-org/pull/56

defunctzombie, 2022-07-05:
@jtbandes Did these packages get published? Some syncs have happened.

jtbandes, 2022-07-06:
I think we might be waiting for one more noetic sync (https://discourse.ros.org/t/preparing-for-foxy-sync-2022-07-01/26297). But this is pretty much done. Keep in mind when we land #17 the new msgs will not be available immediately for all of the distros.

closed 2022-07-11 by jtbandes