foxglove/foxglove-sdk #20: Point2 vs Point2D?

amacneil, 2022-05-07:
These feel unnecessarily abbreviated to me:

* Point2
* Point3
* Vector2
* Vector3

Is there some prior art we are copying for consistency? Or should we add an extra `D` to each of them?

jhurliman, 2022-05-07:
In Eigen, a `Vector2d` is two 64-bit doubles and a `Vector2f` is two 32-bit floats. Is the idea of adding a D to signify that these are all double precision?

jtbandes, 2022-05-07:
> Is the idea of adding a D to signify that these are all double precision?

I’m pretty sure Adrian meant the D to stand for “dimensional”?

Thought experiment: if we wanted to add an integer version of a point (for pixel/grid coordinates), what would we call it?

amacneil, 2022-05-07:
If there is some prior art we are copying that's fine... just looks weird to my untrained eye.

I notice that ROS has both `Pose2D` and `Vector3`, so I guess that's not helpful.

> In Eigen, a Vector2d is two 64-bit doubles and a Vector2f is two 32-bit floats. Is the idea of adding a D to signify that these are all double precision?

😭 that's terrible - if I saw `Vector2d` out of context I would expect it to be a two dimensional vector, not a double precision 3 dimensional vector

> Thought experiment: if we wanted to add an integer version of a point (for pixel/grid coordinates), what would we call it?

`Point2I` 🙈

jhurliman, 2022-05-07:
Vector3 prior art: https://docs.unity3d.com/ScriptReference/Vector3.html

amacneil, 2022-05-08:
Looking at the Unity docs you linked, `Vector3` is kind of an outlier...

[image]
[image]
[image]

From googling, Vector3 does seem pretty common across many languages/frameworks. But those same frameworks also use 2D/3D for other types. All hope is gone.

closed 2022-05-08 by amacneil