foxglove/foxglove-sdk #25: Icon markers in images + 3D

amacneil, 2022-05-12:
Would be awesome to support rendering "icon" markers (image + 3d), like webviz demo mode supports. We could either ship with some standard icons, or let you reference a URL or something.

_Originally posted by @amacneil in https://github.com/foxglove/message-schemas/issues/24#issuecomment-1125260438_

jtbandes, 2022-05-18:
FWIW here are the icons bundled with webviz:

https://github.com/cruise-automation/webviz/blob/d82468c05fb2d3aeca670ff37999faeafa4d7ffc/packages/webviz-core/src/panels/ThreeDimensionalViz/constants.js#L41-L75

This feature isn't related to demo mode in webviz.

jhurliman, 2022-05-18:
Referencing a URL would be a great start. We could provide some `data:` URL examples of base64-encoded SVGs.

Implementing this in the New3D panel should be straightforward, there is code to render arbitrary canvases to sprites.