foxglove/foxglove-sdk #32: Add supported formats in description for RawImage and CompressedImage

esthersweon, 2022-05-20:
Updated in website docs here: https://foxglove.dev/docs/studio/panels/image

amacneil, 2022-11-03:
Another customer asked this question today. I think we should focus on adding this to the schema docs.

I understand that in a pure theoretical world the schemas are separate from the panels that interpret them, but in the real world the only place these schemas are used is in our app, and the schema docs are the place you are looking when you try to figure out what to put in each field.

cc @esthersweon @defunctzombie

jtbandes, 2022-11-03:
There are still two ways we could implement this:
1. We could add the information to the schema itself (e.g. as a description on the `format` field) and it would appear in the generated code comments/docs.
2. We could keep the information in the website repo, and insert/append it when we generate the docs for the relevant schemas. (The same information could be auto-inserted on the 3D panel page as well as the schema pages.)

amacneil, 2022-11-04:
There is a third option: you could add a new field to the source definition in this repo, but only use that data in the docs generation and not schema generation.

I don't feel strongly either way. Adding it to the website docs is the highest priority.

I can see arguments for and against also adding it to the schema comments:
- Pro: Easy for people to access while they are reading the source (especially if they copy the .proto into their own repo)
- Con: Might get out of date if/when we add new formats (especially if they copy the .proto into their own repo)

closed 2022-11-16 by amacneil