foxglove/foxglove-sdk #50: Protobuf schema definition for `Marker` is missing?

felixf4xu, 2022-07-19:
Hi,

I'm using websocket to display online data in protobuf format.

I want to display some "bounding box" like cubes, now I see in "3D" pane and its `Marker` schema : https://foxglove.dev/docs/studio/panels/3d#marker

But I don't see protobuf definition for it (only ros1 and ros2). I'm not sure if it's missing or by design.

Do I have to use `visualization_msgs/msg/Marker` in this case? Any document/sample to create a `visualization_msgs/msg/Marker` in python?

Thanks.

defunctzombie, 2022-07-19:
Hi. We are actively working on this here: https://github.com/foxglove/schemas/pull/17 we aren't ready to publish these and they aren't yet supported in Studio but thats the direction we are going. I welcome you to leave your feedback on that PR if you think it meets or doesn't meet your needs.

felixf4xu, 2022-07-19:
Yes I see that, thanks.

Any update on the review process since last commit was on May 20? or Foxglove Studio needs to be ready first?

amacneil, 2022-07-19:
Yes, it is waiting for us to support in Foxglove Studio before we merge the PR (probably in the next few weeks).

In the meantime, as a workaround you can use the ROS messages converted to Protobuf on [this branch](https://github.com/foxglove/schemas/tree/converted-ros-msgs/proto/ros/visualization_msgs).

felixf4xu, 2022-07-30:
Hi @amacneil , I would like to confirm that if I use the protobuf definitions mentioned in your link, which online connection type (as in https://foxglove.dev/images/docs/studio/connection/live-connection/rosbridge-dialog.webp ) should I choose?

amacneil, 2022-07-30:
@felixf4xu yes that is correct, you should use the protobuf messages with `Foxglove WebSocket`.

You can find example code here: https://github.com/foxglove/ws-protocol

Make sure you set the `schemaName` correctly (e.g. `ros.visualization_msgs.MarkerArray` so that Foxglove Studio recognizes it).

closed 2022-08-30 by jtbandes

felixf4xu, 2022-09-07:
@jtbandes Do you know from which version the updated doc of Markers (https://foxglove.dev/docs/studio/panels/3d#marker) will appear on the docs site? Currently it's still missing.

jtbandes, 2022-09-07:
It will be updated after Studio fully supports all the new types. Likely within the next 4 weeks.