foxglove/foxglove-sdk #59: Add `frame_id` to image types

jtbandes, 2022-08-24:
Image and CompressedImage need `frame_id` if we are going to display them in the 3D panel. CameraCalibration may need it as well if we want to use `frame_id` to confirm that the calibration topic corresponds to the image topic.

jhurliman, 2022-08-24:
CameraCalibration is an independently visualizable topic in the 3D panel, so it needs frame_id as well.

jtbandes, 2022-09-06:
@jhurliman actually could you confirm whether frame_id is needed for Image topics in order to display them? If a CameraCalibration is required to display an Image then it seems like the Image itself may not need a frame_id (or I'm not sure what the benefit would be).

jhurliman, 2022-09-06:
@jtbandes `Image` can only be rendered in 3D space with a corresponding `CameraCalibration`. If the frame_id is not being used to match topics together, then it doesn't serve any other purpose. The matching use case may be critical though, because even if Studio doesn't automatically match based on frame_id a human user might need it. Without it, you have to rely on topic name matching (either as an algorithm in Studio or heuristically as a human operator) to pair topics together.

jtbandes, 2022-09-06:
Good point, if we don't have frame_id in the image, then we are forcing users to do something with the topic name, which we generally don't do otherwise.

PR #61 (2022-09-07): add frame_id to RawImage, CompressedImage, and CameraCalibration
**Public-Facing Changes**
Added `frame_id` to RawImage, CompressedImage, and CameraCalibration

**Description**
Closes #59 
Unblocks https://github.com/foxglove/studio/issues/4323
commit b6b7350a (2022-09-07): add frame_id to RawImage, CompressedImage, and CameraCalibration (#61)

**Public-Facing Changes**
Added `frame_id` to RawImage, CompressedImage, and CameraCalibration

**Description**
Closes #59 
Unblocks https://github.com/foxglove/studio/issues/4323
closed 2022-09-07 by jtbandes