foxglove/foxglove-sdk #140: Cannot make flat arrows

jhurliman, 2024-02-20:
**Description**

With ROS arrow markers, you can set `size` to `{1, 1, 0.01}` to get a flat arrow. There doesn't seem to be any way to replicate this with SceneEntity arrows.

- Version: latest
- Platform: all

**Steps To Reproduce**
* Fail to make a flat SceneEntity arrow

**Expected Behavior**
* Succeed in making a flat SceneEntity arrow

defunctzombie, 2024-02-20:
Are `shaft_length` and `shaft_diameter` https://docs.foxglove.dev/docs/visualization/message-schemas/arrow-primitive/ not sufficient?

jhurliman, 2024-02-20:
Unfortunately, no combination of those two values will produce an arrow that looks like this:

[image]

jhurliman, 2024-02-20:
It's a random image I found online that shows a flat arrow, in the same style that the ROS arrow marker can produce. This type of style cannot be replicated with an `ArrowPrimitive` because you can only control head/shaft length and diameter, you can't resize per 3d axis.

defunctzombie, 2024-02-20:
> It's a random image I found online that shows a flat arrow, in the same style that the ROS arrow marker can produce. This type of style cannot be replicated with an `ArrowPrimitive` because you can only control head/shaft length and diameter, you can't resize per 3d axis.

Gotcha. Thanks for the additional context. I'm gonna move this to feature cause our current arrow primitive is designed for making volumetric? arrows.