foxglove/foxglove-sdk #611: Foxglove Bridge Rejected client

eladpar, 2025-08-21:
**Description**
Im launching foxglove fot he first time on my remote machine and i could not connect to the local ros2 topics
on my foxglove bridge i get this:
```bash
[foxglove_bridge-1] [INFO] [1755762312.564004272] [foxglove_bridge]: [WS] Rejecting client 127.0.0.1:55680 which did not declare support for subprotocol foxglove.websocket.v1
[foxglove_bridge-1] [ERROR] [1755762312.564096338] [foxglove_bridge]: [WS] Handshake ended with HTTP error: 400
```

- Version:2.34.0
- Platform: amd64

**Steps To Reproduce**
download fox glove and ros2 launch foxglove bridge

**Expected Behavior**
to be able to see ros2 topic on screen

amacneil, 2025-08-21:
Hi @eladpar, that’s unusual that it works remotely and not localhost.

Would you mind building the latest version from source as documented here to see if the issue still exists?

https://github.com/foxglove/foxglove-sdk/tree/main/ros/src/foxglove_bridge#build-from-source

eladpar, 2025-08-21:
> Hi [@eladpar](https://github.com/eladpar), that’s unusual that it works remotely and not localhost.
>
> Would you mind building the latest version from source as documented here to see if the issue [...]

So no more Error messages with the bridge built from source.
I can only see on the local foxglove instance " check that you local connection at ws:localhost:..." is reachable,
but still not working...
can it be a firewall issue somehow although its localhost?

ericmlujan, 2025-08-21:
That is indeed strange, I'd expect localhost connections to work (I also pulled and rebuilt Foxglove Bridge from source, and loaded your version of the Foxglove app on Ubuntu 24.04, and had no issues streaming local topics), barring any network/security/firewall configuration that would block the connection. Are you seeing any output other than the "server listening" message coming from your Foxglove Bridge process? For more granular log messages to help with debugging, you can `ros2 launch foxglove_bridge foxglove_bridge_launch.xml debug:=true`.

Another thing worth trying is seeing if you can connect to your local bridge from https://app.foxglove.dev in your web browser. If you can't, please share any error output from the Chrome developer console with us, so we can see if anything stands out as unusual.

eladpar, 2025-08-25:
> That is indeed strange, I'd expect localhost connections to work (I also pulled and rebuilt Foxglove Bridge from source, and loaded your version of the Foxglove app on Ubuntu 24.04, and had no [...]

so somehow when I used the debug true the local foxglvoe didn't work and I got the same messages,
but then a day after, I tried the website and It worked and now the local foxglove works too.
So a bit weird but it seems to work for now

ericmlujan, 2025-08-27:
Glad to hear things are resolved! Given the temporary nature of the problem, and a lack of messages on the bridge side pointing to an error setting up a connection, I think it's best to close this one for now. If the problem recurs, feel free to reopen and provide additional information about your environment, log outputs, and reproducibility steps.

closed 2025-08-29 by ericmlujan